CarlosValente93092/Projeto_Industrial #55: Ver custo dos materiais necessários

CarlosValente93092, 2022-11-05:


dFernandes15, 2022-11-07:
Lista de material:
- Mosfet VN2222LL: 1,09€ 
         https://www.tme.eu/pt/details/vn2222ll-g/transistores-com-canal-n-tht/microchip-supertex/

- Opto-coupler HCPL-2611: 1,50€
         https://www.tme.eu/pt/details/hcpl2611/acopladores-oticos-saida-digital/onsemi/

dFernandes15, 2022-11-07:
Ter em conta que estes foram os componentes utilizados na documentação fornecida pela empresa. Não devemos descartar usar modelos de componentes com especificações semelhantes mas que estejam disponiveis no DETI

CarlosValente93092, 2022-11-08:
Na mouser parece mais barato:
VN2222LL-G : https://pt.mouser.com/ProductDetail/Microchip-Technology-Atmel/VN2222LL-G?qs=H6g4Fy%252BsDGemwmUTAFka1Q%3D%3D

dFernandes15, 2022-11-08:
Visto que ainda não é certo o uso da Orange Pi, vou supor por enquanto o uso da Raspberry Pi 3 model b+ com modulo WiFi e bluetooth 4.2
- https://mauser.pt/catalog/product_info.php?cPath=1667_2620_1672&products_id=096-5700

closed 2022-11-16 by CarlosValente93092