CarlosValente93092/Projeto_Industrial #66: Ver circuito de RS-485

CarlosValente93092, 2022-11-11:


dFernandes15, 2022-11-13:
SN75176BP é o modelo do transceiver usado na documentação recebida.
(Existe também o MAX485 mas é mais caro.)

https://pt.mouser.com/ProductDetail/Texas-Instruments/SN75176BP?qs=FetendrIl6uStzmaeBLInA%3D%3D

dFernandes15, 2022-11-14:
Alguns links com informação acerca do assunto:
- https://www.renesas.com/eu/en/document/whp/rs-485-transceiver-tutorial?language=en
- https://www.microchip.com/forums/m1091421.aspx

Este último é bastante útil pois trata-se de um forúm com perguntas e respostas interessantes.

dFernandes15, 2022-11-14:
O modelo que temos SN75176B é adequado para o projeto e apresenta todos os requisitos de acordo com o primeiro documento do comentário anterior https://www.renesas.com/eu/en/document/whp/rs-485-transceiver-tutorial?language=en

closed 2022-11-16 by CarlosValente93092